LakeMaps/Dartmouth #1: Motherboard Design

FifoIronton, 2016-07-27:
We need to map out what we want on this board.
This board is pretty much a header for the Raspberry Pi.
Needs:
- Power rails, 
- 12V to 3V, 12V to 5V buck convertors
- Standard mother to daughter 8 pin JST terminals ( 2 GPIO, 3.3V, 5V, GND, SDA&SCL, something else)
- USB Bus?
- Logo

We should use this issue to develop a discussion of the components.

arandell93, 2016-07-27:
I don't think we need a USB bus. We can't get a USB bus off of the headers on the Pi anyway.

We need to have no connects to all of the Pi's voltage outputs, since we're going to be bringing in our own power from our DC/DC converters to all the devices. However, all of the grounds will be common (unless there is a reason not to, but I can't imagine having a noisy ground in this system).

For the pins that aren't scheduled to be used for daughterboard connections, we should break them out onto similar connectors anyway in case we want to use them. For example, the SPI bus on the Pi could be brought out to an 8pin JST with 5V, 3.3V, GND and a couple no connects.

arandell93, 2016-07-27:
Not sure if JST is the best to use, I think putting the pins in is difficult. 
What are your thoughts on using something like CAT5/RJ45? They're easy to use, the crimp tools are cheap and I have 185 feet of cable already.
I know it's kinda gross because we'd be using a standardized port for something non-standard, but there are a good few pros.
Thoughts?

FifoIronton, 2016-07-27:
I'll be honest, I just kind of assumed that this connector was JST : http://www.digikey.com/product-search/en?mpart=71600-108LF&vendor=609 . These interface with 10mil pin headers, which are nice.

Turns out I was wrong. If you have CAT5 cable and the skills to terminate, although it hurts me I can do it, because cheap solutions sound like a good idea

arandell93, 2016-07-27:
I haven't seen those before haha. JST is super small.
It hurts me too but I think it's a good decision in terms of ease of manufacture and costs. RJ45 it is!

arandell93, 2016-08-11:
Preliminary _in progress_ design
[image]

arandell93, 2016-08-11:
Markup as submitted to @FifoIronton on 2016-08-10 22:45
[image]

closed 2016-11-06 by FifoIronton