LakeMaps/Dartmouth #2: Radio Module Daughterboard

FifoIronton, 2016-07-27:
Need to create a radio module breakout board for RFM9XW. We want:
- RPSMA output
- SPI to microUSB port
- 3.3V linear regulator

arandell93, 2016-07-27:
Since we need two modules (one on the boat and one on shore), I think we're best making them identical.

Note: We are going to be bringing ONLY a USB cable to these modules. Power will be in the form of 5V via USB, so a USB hub is required to provide enough current.

Any SPI to USB converter chip we get will have to be capable of at least 300 kbit/s

arandell93, 2016-07-27:
I'm gonna go ahead and make this priority ahead of the motherboard, as I feel this will take the most work to get sorted (config, testing, antenna messing), and it is independent of the motherboard so the latter work can be parallel to the development of the other boards.

arandell93, 2016-07-27:
Potential SPI to USB converter:
https://www.digikey.ca/product-detail/en/microchip-technology/MCP2210-I-SS/MCP2210-I-SS-ND/2835087
Also available in SMD, but I think this package would be easier to solder to.

FifoIronton, 2016-07-27:
We wouldn't be able to use the alternate package that they have, as the pads are underneath the chip on that package.

Speaking of that, we should probably aim to get several of the boards done so I can solder and verify them at Solace - the pins on the chip you've shared are only separated by about a millimeter, and Solace much smaller soldering irons than I have access to at MUN. 
I'm not sure if Solace'll let me borrow a key for after hours access during the semester, but I will ask about it closer to the end of my work term.

arandell93, 2016-07-27:
Yeah that's what I'm thinking. I'll create a tracking issue for that.

FifoIronton, 2016-08-01:
First draft of schematic fleshed out as of 87f1c00 

If you think this looks appropriate I will continue working towards finishing layout ASAP
[image]

arandell93, 2016-08-01:
Are we using any of the IO on the radio module? I see you've got them all connected to the USB-SPI bridge GPIO, but do we know how to get at the GPIO in software via the USB interface?

FifoIronton, 2016-08-01:
I'll be honest, I haven't actually read the full data sheets for both of them. I'm like 80% sure that the GPIO are controlled by the USB interface, as they'd be needed as for Slave Select (SS) pins for the SPI interface (assuming you're not doing some complex Daisy Chaining thing). 
Based on the LoRa balloon system I've shown you before, the guy connects some of the IOs on that module for selecting options or something in his [software](https://github.com/daveake/LoRaArduinoSerial/blob/master/LoRaArduinoSerial.ino). It's another thing I haven't actually read, but I want to figure out what's going on there

FifoIronton, 2016-08-07:
Next time let's get the SOIC footprint instead of the SSOP footprint. Wider lead spacing, which is good for me

FifoIronton, 2016-08-17:
[image]
Things to be done:
- Change hole sizes
- Put a fun label on?
- make sure it doesn't suck

arandell93, 2016-08-17:
- Am I correct in understanding that the MCP2210 is underneath the RFM69HCW where the MCP is surface mount on one side and the RFM is through-hole mounted on the other?
 - Side note: We will need pins (what spacing) to mount the RFM through-hole.
- Power LED resistor should be 33 Ohms.

FifoIronton, 2016-10-25:
Rev1 Schematic of this board
[image]

closed 2016-11-06 by FifoIronton