LakeMaps/Dartmouth #3: Choose kooky naming scheme for boards

FifoIronton, 2016-07-27:
When I worked at a big company, families of boards that they designed all had internal names based on a   set theme. For example, in one family, all cards were named after stars, and in another family all of the cards were named after characters from a movie.

I think we should emulate this on our project, mostly so that everything sounds a little sillier when we talk about it. Also, I'm jealous that the control software repository gets to be called "boat", while I have a repository called "PCB" currently.

Some options for a board naming theme:
- YuGiOh Cards in a 2011 era Frog deck, or maybe an Elemental Hero Absolute Zero Deck
- Names of lakes geographically close to my family home in Dartmouth
- Different names of blue Crayola crayons, with the motherboard being named "Blue"
- My favourite flavours of herbal tea

What do you guys think?

arandell93, 2016-07-27:
I like the names of ponds & puddles close to your family home in Dartmouth.
We can call the motherboard Dartmouth then?

I also like the blue crayola crayon idea.

FifoIronton, 2016-07-27:
Dartmouth it is then, I'll implement the changes when I get back home

arandell93, 2016-07-27:
Close the issue once you've implemented the naming scheme.

closed 2016-07-27 by FifoIronton