LakeMaps/Dartmouth #5: Dartmouth Daughterboard Interface

arandell93, 2016-08-03:
The plan is to use a USB Hub and multiple USB to UART converters in order to achieve the required number of serial ports.
We require the following hardware for this implementation:
- USB-A to USB Micro-B cable
- USB Micro-B port
- USB Hub IC ([TUSB2036](http://www.ti.com/product/tusb2036), [USB2502](http://ww1.microchip.com/downloads/en/DeviceDoc/2502.pdf) or similar)
- External Oscillator for above (depending on IC selection)
- USB to UART IC (Preferably multiple in one IC like the quad [FT4232H](http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT4232H.pdf))

@FifoIronton have a quick google and let me know what you think. The USB Hub ICs are a lot of pins, so soldering could get quite difficult.
It's also very important that we completely understand the interfacing requirements prior to purchasing. For example, it looks like the [FT4232H](http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT4232H.pdf) has other connections besides USB, and we don't want things to get complicated.
We always have the 'dumb' alternative of buying a USB hub and connecting several USB cables with single USB-Serial converters on it.

FifoIronton, 2016-08-17:
[image]
[image]
[image]
Dump of Dartmouth Images

FifoIronton, 2016-08-17:
Still to do 
- Put some lettering in front copper layer on board
- Change all drill holes to holes APC will do for free
- Make sure it doesn't suck

arandell93, 2016-08-17:
- 150 Ohm resistor is not appropriate for 5V and 3.3V LED. 3.3V resistor should be lower. I assume we're using 10mA LEDs with a 3V drop so the correct values are 200 (210 standard) for 5V and 30 (33 standard). Doesn't affect the board itself but note the change in the next board revision.
- There isn't a requirement to bring 3.3V from our DC/DC converter to the Pi's GPIO because the Pi generates its own 3.3 V. The fuse in this location is also consequently not needed.
- The intention of the fuses on the board is to protect the line (12 V) side of the dc/dc converters individually from short circuiting, as well as one on the input to the Pi to make up for us bypassing it and going to the GPIO. The line side fuses do not currently exist.
 - If it would be easier, we can use the existing 3.3V fuse as the load side of the 3.3V dc/dc converter and have all 3.3V loads running through it (except the unnecessary Pi connection as noted above).
 - We will still need to add a fuse for the 5V as well. If we do load side for the 3.3V we can do load side for the 5V (aside from the Pi which still needs a dedicated fuse). However, my preference would be to re-purpose the existing 3.3V pi fuse into a fuse protecting the line (12V) of the 3.3V and to add another fuse on the line (12V) side of the 5V converter.
 - Both options require the addition of one fuse to the board for a total of 3.

closed 2016-11-06 by FifoIronton