LakeMaps/Dartmouth #6: Create logo and decide on vendor for PCB manufacturing

FifoIronton, 2016-08-04:
Somebody's gotta create a Black/White logo for me to put on the silkscreen of the boards. You know it will look way cooler if we do that (assuming we're getting the solder mask and silkscreen on our boards).

If nobody else does it, I'm going to make something stupid, so be warned

arandell93, 2016-08-04:
Yeah I agree we need a logo haha. I've been looking at some lakes with a mountain background kinda things but I'm awful at making creative things.

What are you thinking for a vendor?

FifoIronton, 2016-08-04:
I spent a bit of time looking today, and for our order quantities we're kinda out of luck for getting a cheap product with low lead time. We could, if we commit to single layer boards with only surface mount components, we can [etch our own boards](http://makezine.com/projects/cheap-friendly-and-precise-pcb-etching/), which means that we have to jump through a bunch of design hoops and spend time making them, but we end up saving a fair bit of money for a product similar to what we'd get from APC.

If we decide against that, it appears that for extremely low numbers of boards APC is a good company to work with. Otherwise, we can go through China with a company like [this one ](https://www.seeedstudio.com/fusion_pcb.html)my friend recommended, even though they require a minimum of 5 boards (that being said, if you want tiny boards they'll be like $2USD each + shipping).

You can also look at this [list here](http://digitaldiy.io/general-electronics/321-pcb-fabrication-sources.html#.V6ONqbuAOko) that walks through a bunch of places (mostly for US customers) and outlines their prices

cgregory52, 2016-08-04:
I know very little about potential issues with PCB printing so take my opinion with a grain of salt. Anything handmade has a high likelihood of failure so count on double to tripled costs for when the part inevitably breaks and we need a new one. In my mind, the small cost savings per board are a negligible bonus here so the decision should be made based on which design is better/easier. If we're jumping through design hoops, potentially creating a lower quality product on our own with a high rate of failure I do not see this as being worthwhile. Printed boards remove a large number of potential failure points from debug considerations as well. We wouldn't necessarily need to look at every PCB solder joint as a point of failure each time something stops working. Time and money are both resources and for the potential time savings alone, I think the additional monetary cost is justified.

FifoIronton, 2016-08-04:
With that in mind, I will proceed with my layouts using 2 layer boards, because they are much easier to design than single layer boards. I just wanted to present the most cost effective option, while also showing the alternatives.

The next question is just how much we're willing to spend on PCBs and how important it is to us that they're done quickly. APC is definitely the most convenient option because we've worked with them before, they're based in Canada, and they have really low lead times, but for the batches we're doing they're pretty pricey. If we go through China it's hard to get small batches, but we can get big batches for cheaper than APC will give us the small batches, but we gotta wait.

arandell93, 2016-08-05:
I'm definitely in favour of getting them done by someone.
I think if we can get our quantities high enough we should go with one of the cheaper Chinese options provided the lead times aren't insane.

closed 2016-11-06 by FifoIronton