LakeMaps/Dartmouth #7: GPS PCB Design

arandell93, 2016-08-08:
The PCB for the GPS needs the following:

- 6 pin connector for the GPS module
- 5 V Power
- 5 V TTL UART (TX and RX)
- Associated passive components (Caps, etc)
- Power LED

The GPS module is an [EM 406](https://www.sparkfun.com/datasheets/GPS/EM-406%20Product_Guide1.pdf).

arandell93, 2016-08-11:
Initial Design:
[image]

arandell93, 2016-08-11:
### Requires
- Ethernet Port Shield (9) should be grounded.

FifoIronton, 2016-08-17:
[image]
To Do:
-Change hole diameters to the free holes from APC
-Make the layout eaier to read, even though there won't be any silk screen
-Make sure it doesn't suck

arandell93, 2016-08-17:
- LED resistor should be 210 ohms.
- Confirm GPS port is correct type (JST SH Vertical [or Horizontal] 6-Pin)

closed 2016-11-06 by FifoIronton