LakeMaps/Dartmouth #8: Sonar PCB Design

arandell93, 2016-08-08:
The PCB to integrate with the Sonar unit requires the following:
- 12 V
- RS-422 to TTL converter (Such as [3179E](http://www.intersil.com/content/dam/Intersil/documents/isl3/isl3179e-80e.pdf) or [7306](http://www.analog.com/media/en/technical-documentation/data-sheets/AD7306.pdf)). These will convert the 422 from the NMEA 0183 sentences the sonar outputs to TTL for interpretation by our controller.
- Caps as required
- A 7 pin connector (Need to know wire size on transducer cable in order to size connector type)
- In-line fuse on 12 V input (2A)

Note: No connection on the ACC ON pin.
[image]

FifoIronton, 2016-08-09:
According to this [TI application note](http://www.ti.com/lit/an/snla041a/snla041a.pdf), there is no need for a converter, as RS-422 can drive a TTL receiver. Look at page 5.

arandell93, 2016-08-09:
Interesting...
Seems legit though. As long as the TTL input on the chip we select for our UART can tolerate the voltage levels we should be good to go! I'd say anything that's 13 V tolerant on the input would be good.
Let's allocate for a voltage divider on the RX of the UART chip to drop the voltage to something safer just in case the TTL isn't tolerant of the voltages. If we don't need it we can jump it.

arandell93, 2016-08-11:
Preliminary _in progress_ design
[image]

arandell93, 2016-08-11:
Initial Design:
[image]

arandell93, 2016-08-11:
### Requires
- Power LED (From 12 V)
- 2A Inline Fuse (20x5mm horizontal fuse holder)

FifoIronton, 2016-08-17:
We still haven't decided on a termination for the sonar wire. Not sure what to do, but layout will be easy once we decide. @arandell93 or @michaelabarnes , you guys know sonars, what is the standard termination? If I don't get a response by Wednesday night I'll assume something simple with 10mil spacing

arandell93, 2016-08-17:
The intelliducer comes with non-terminated ends. Let's do something simple and easy to repair/replace. I'm tempted to do something dumb like ring terminals. You often see an XLR looking connector but that's complicated and expensive.
If I knew how big the wires were I'd pick a c-grid and be done with it, but I'm concerned that the wires are going to be like 18 gauge and then we'd be out of luck as far as a c-grid goes. We could use minifit connectors (There are some at EER that will never again be used because I hid them all away and they're only rated for 6A and are really big). They'd work well enough for us.
Thoughts?

FifoIronton, 2016-08-17:
From looking "Minifit connector" up on google, it looks like they still require you to put crimp connectors on the wires, are they bigger crimps than in C-Grids?

arandell93, 2016-08-17:
Yeah the minifit will take up to a 16 I think, depending on the crimp pins you buy. Definitely big enough for our purposes. You can also get right-angle and vertical through-hole connectors, in fact we may have some available to us. Worth looking at the shop.

closed 2016-11-06 by FifoIronton