LakeMaps/Dartmouth #11: Options for Radio Communication

FifoIronton, 2016-11-06:
Relooking at how to attach the RFM69HCW Boards to our system means we need to evaluate different options. I'm going to present all the options I can see, with comments on how much I like them. The major issue we've faced with the RFM Modules is that they have 3.3V logic levels, and we're trying to throw 5V into them, which is bad.

* Buy [RFM Module](http://www.digikey.ca/product-search/en?mpart=COM-13909&vendor=1568), solder some resistor dividers onto a protoboard for the level shifting, and also solder jumpers from the module to the protoboard. Add some headers, connect to the Arduino Unos we have, and we should be good.
This idea is the cheapest option, and has a high chance of working with very minimal changes to our system. The only issue that we may run into is the stray capacitance of the resistors, which could mess with the SPI signals... Doubt it, but it's a possibility. As we are using physical components to step down the voltages, we are limited in how we use the IOs with this, but I doubt we'll really use those on the radio. For the sake of a few extra cents, we could also try using [through-hole level shifters](http://www.digikey.ca/product-detail/en/texas-instruments/CD4504BE/296-3527-5-ND/376626) to save space and make it look a little nicer 
  * Materials: 2x [RFM69HCW Modules](http://www.digikey.ca/product-search/en?mpart=COM-13909&vendor=1568) ($7.14 each), protoboards, resistors, etc, from MUN.
  * Cost: $14.28 + Tax, Shipping
  
* Buy a [radio breakout board with level shifters built in](http://www.digikey.ca/product-detail/en/adafruit-industries-llc/3070/1528-1665-ND/6005355). Although a little more expensive, this solution has a couple of benefits over the previous option. We can easily insert this breakout board into a breadboard for prototyping, as we can solder 100mil headers onto this board, and it runs directly off of 5V logic without the need for our own resistor divider circuits.
  * Materials: 2x [Adafruit 3070 breakout boards](http://www.digikey.ca/product-detail/en/adafruit-industries-llc/3070/1528-1665-ND/6005355) ($14.36 each), breadboard, 100mil headers ([10 pos](http://www.digikey.ca/product-detail/en/sullins-connector-solutions/PREC009SAAN-RC/S1012EC-09-ND/2774845) and [5 pos](http://www.digikey.ca/product-detail/en/harwin-inc/M20-9990546/952-2268-ND/3728232))
  * Cost: $29.72 + tax, Shipping

* Buy our own 3.3V [Arduino Pro Micro boards](http://www.digikey.ca/product-search/en?mpart=DEV-12587&vendor=1568), hook them up to RFM69HCW modules. Pretty simple, but more expensive than buying the breakout boards with 3.3V built in... but then we own 3.3V microcontrollers, freeing up our Unos for other tasks and allowing us to upgrade to the LORA boards in the future without having to pay for level shifting boards again. I like this option, as it is more compact than using an UNO, but am a little hesitant based on the price. There are other cheaper options involving Arduino Pro Mini boards, and the magic of a USB to serial converter, we we can use to bring the cost down, if this sounds like a good option.
  * Materials: 2x [RFM69HCW Modules](http://www.digikey.ca/product-search/en?mpart=COM-13909&vendor=1568) ($7.14 each), 2x [Arduino Pro Micros](http://www.digikey.ca/product-search/en?mpart=DEV-12587&vendor=1568) ($29.79 each)
  * Cost: $73.86 + Taxes, shipping
* Buy Adafruit's [RFM69HCW Feather Board](http://www.digikey.ca/product-detail/en/adafruit-industries-llc/3076/1528-1663-ND/5983112). These are as expensive as the previous solution, but also idiotproof. They've got everything built into one package, so you know that you're getting what you're looking for; however, if we want to switch radios in the future then we have to pay for all the extra microcontroller stuff again.
  * Materials: 2x [RFM69HCW Feather Boards](http://www.digikey.ca/product-detail/en/adafruit-industries-llc/3076/1528-1663-ND/5983112) ($36.01 each)
  * Cost: $72.02 + Taxes, shipping

Thoughts?

arandell93, 2016-11-06:
A few comments:

- You left out the RFM69HCW modules in the breakout board option. Do they come preinstalled?
- I like that the breakout board has the ability to solder on an SMA connector directly. While we could do it with our own protoboard it would be a bit messy. Using the feather would require us to use a uFL connector and then an adapter to SMA which I'm not a fan of.

I like the breakout board option since it gives us easy prototyping ability and is still relatively low cost.

FifoIronton, 2016-11-06:
I'm 95% sure that the breakout boards come with the radio modules pre-installed, based on the [description given on Adafruit](https://www.adafruit.com/product/3070), the manufacturer. Sounds like a good solution for now, we should put in an order after figuring out if there's anything else from Digikey we should get.

arandell93, 2016-11-06:
Yeah it appears they do have the boards on them.
What else might we need?

FifoIronton, 2016-11-07:
I guess it depends on if we want to pick up antennas. It's pretty early for that now, but we could pick them up. Other than that, all I see us needing are the boards, along with some headers. I have two breadboards, but if you wanted [more](http://www.digikey.ca/product-detail/en/bud-industries/BB-32621/377-2094-ND/4156445) we could order them. I don't recommend it, as there's no time crunch and they're [cheaper out of china](https://www.amazon.ca/COLEMETER-Solderless-Breadboard-Points-Contacts/dp/B00CGVFPRY/ref=sr_1_2?ie=UTF8&qid=1478479307&sr=8-2&keywords=breadboard).
... Beyond that, I can't really think of anything.

arandell93, 2016-11-07:
I agree we don't need to buy an antenna or breadboards.
The boards come with headers, don't they?

FifoIronton, 2016-11-07:
I think so, but it only mentions it on the Adafruit page, not on Digikey, and headers for the each of the boards are 50¢, so I think we're better off splurging.

arandell93, 2016-11-07:
Good point. We'll order a few. We'll probably need some at some point for other things anyway.

arandell93, 2016-11-08:
I'm going to place the order for the following today. Confirm that it looks good to you. please @FifoIronton.

QTY 2:
[RFM69HCW Breakout Board from Adafruit](http://www.digikey.ca/scripts/DkSearch/dksus.dll?Detail&itemSeq=211263501&uq=636141826849499853)

QTY 2:
[40 Position, 0.1" Male Breakway Headers](http://www.digikey.ca/scripts/DkSearch/dksus.dll?Detail&itemSeq=211263745&uq=636141828558100696)

FifoIronton, 2016-11-08:
The links you provided in your comment don't work for me, but sounds good to me, regardless.

FifoIronton, 2016-11-08:
I can't quite figure out which issue I should mention this on, so I'll also mention it here - we need another Arduino if we're going to have a separate Hangashore comms Arduino, Pudding comms Arduino, and Pudding motor control Arduino. We can buy a [legit one](http://www.digikey.ca/product-detail/en/arduino/A000066/1050-1024-ND/2784006) from Digikey for >$30, or we can be bad people and buy a cheap clone (something like [this](https://www.amazon.ca/LANDZO-Arduino-Atmega328-Maker-Learner/dp/B01IGQ38DU/ref=sr_1_20?s=electronics&ie=UTF8&qid=1478618420&sr=1-20&keywords=uno+r3)), understanding that there will be a little more lead time.

arandell93, 2016-11-08:
Good point. I've got a spare arduino somewhere at my house, but I'm not sure whether it's an uno or a micro. It's from my senior design project. I'd prefer not to buy anything further for Rev 1 so we'll make due with that, or borrow one of EER's Megas or something.

closed 2017-03-07 by arandell93