LakeMaps/Dartmouth #12: Standardization of Electronics

arandell93, 2017-03-06:
I am in the process of reviewing our current approach to hardware design and believe that it is in our best interest to begin standardizing certain aspects of our system. This includes things like components, colours, and other design motifs. 

I've added a folder to the Google Drive called design resources which we can use to hold appropriate best practices and standards. We should use these where possible to justify our decisions.

This issue is to facilitate the discussion of the standardization in terms of modifications to existing systems and to (as much as possible) avoid issues in the future with the required hardware outlined in the development milestones.

arandell93, 2017-03-07:
Some talking points. @FifoIronton @whymarrh @cgregory52 looking for all of your input and priorities.

## Wire Colour
Wire colouring serves as a method by which a designer communicates the function of a circuit quickly and unambiguously. This is used to increase safety and decrease troubleshooting time. There are various other lesser considerations we may come across and want to apply, but we optimize for the two aforementioned goals.

To what level of specificity do we want to assign wire colours to meanings? Generally, there are several schools of thought in assigning wire colours:

- Voltage-based. This is generally the minimum and is used as a form of safety indication. By looking at a wire you know what voltage would exist if it were energized and you can determine whether or not you should be working on a circuit while it is live. This is common in mixed AC/DC power systems.
- High-Level function. As a step up from voltage-based, you would colour wires that are used for signalling differently than those used for power distribution. This differentiates between power distribution, communications, and input/output. This is common in industrial control applications.
- Low-Level Function. Here, you would employ a large variety of both solid and striped colours in order to assign wires on a nearly one-to-one function basis. This is common in the automotive industry.

Currently on the boat, we have the following functions (at the lowest level):

- +12 VDC
- +5 VDC
- Common (0 VDC, ungrounded)
- PWM controlled motor power (+/- 0-12 VDC)
- UART TX at various baud rates (microcontrollers use 115200, motor driver uses 9600 but we can [change this](https://www.pololu.com/docs/0J29/4.b), GPS uses 4800 but we can change this (lakemaps/boat#40).
- UART RX at various baud rates
- Reset Signal to motor driver
- Error Signal from motor driver
- Reset Signal to wireless module
- SPI (4 wire; CLK, CS, MOSI, MISO) to/from wireless module
- Interrupt from wireless module

## Wire Size

## Wire Routing/Grouping and Component Layout

## Termination Types

## Microcontroller Type

FifoIronton, 2017-03-07:
My opinions on wire colour are to make as many distinctions between signals as is practical. On a previous project that we all worked on, we used as many colours as we could (each DC voltage had a colour, each class of signals had a colour, lines that were easy to confuse like tooling motor voltages were different colours), and it made the mess of wires inside of the robot bearable.

I do not think that we can separate every single function into a different colour line, but if we made the default digital signal colour grey (or whatever we can get cheap), and differentiated signals from there, I think that we could make the system easier to work with

arandell93, 2017-03-07:
I tend to agree, as I knew the guy who came up with the colours for that project and generally think he knew what he was at. However, I feel there were flaws in that system and would like to resolve them here. I feel it was not explicit enough and was over-constrained in some cases, particularly when it came to edge cases (control PWM, tooling motor power).

I'll propose the following classifications for wires. We can add colours later. Input welcome.

- 12 VDC (I propose this include the switched power to motors or similar devices)
- 5 VDC
- 3.3 VDC (Not currently used, but may be worth reserving unless we decide to standardize on 5V and work around requiring transport of 3.3 V via wire)
- Common/Ref
- Asynchronous Serial
- Synchronous Serial
- Digital (including PWM and other switching control signals)
- Analog (Not currently used)

arandell93, 2017-03-07:
##Wire Specs

For wire specs, I think it would be worth standardizing all non-current-carrying (control) conductors to a single type of wire. This would not only include gauge but insulation type and strand count/size. Similarly, we would apply this insulation type and strand size to the power wiring, though it would have a different number of strands and would obviously be larger gauge. We must be able to easily source this wire from somewhere like digikey or newark.

For control conductors, I propose the following:

- #24 AWG
- Type TEW, 600 V, CSA

To be continued

arandell93, 2017-03-08:
Gonna go ahead and say we're gonna have TX and RX be different colours for debugging purposes. I'll edit my comment above.

FifoIronton, 2017-03-08:
@ Wire Spec - I've been asleep for quite a while but... are we using stranded wires for our digital signals? For prototyping, solid core wire generally works much better with breadboards and arduinos and all that, but in my slumber I've really been missing out on what the system looks like at this point

arandell93, 2017-03-08:
> are we using stranded wires for our digital signals
Stranded for everything. Solid is not suitable for much other than prototyping. We will put header pins on all female headers so we can connect using appropriate wiring with terminations.